pez2001/razer_chroma_drivers #133: Razer Ouroboros gaming mouse support

hotice, 2016-05-27:
Would it be possible to add support for Razer Ouroboros mouse? It's a wireless/wired mouse, here's the product page on razerzone.com: http://www.razerzone.com/gaming-mice/razer-ouroboros

ID when used wired: 1532:0032; ``lsusb -v -d 1532:0032`` output: http://pastebin.com/raw/FjkzhM2C

ID when used wireless: 1532:0033; ``lsusb -v -d 1532:0033`` output: http://pastebin.com/raw/dueQU1xu

terrycain, 2016-05-27:
I'm busy working on supporting some other devices. I think I'll need to make a queue for this ha.

The main answer is yes - if you can help with the process. Read [this](https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol), if your able to do that then yes (after I've done other devices).

hotice, 2016-05-31:
That might be too advanced for me but I'll look into it.