pez2001/razer_chroma_drivers #135: keyboard flickers when daemon is initialized on Blackwidow Ultimate 2016

DasHammett, 2016-06-05:
Hi all, 

As soon as I initialize razer_bcd daemon, the backlight flickers at different speeds, making the lightning very annoying. If  I stop razer_bcd the flickering stops.

System: Gentoo Linux - 4.6.0 Kernel

If you need extra information, please let me know.

Thank you

Kind regards,

terrycain, 2016-06-05:
When the daemon is running try running `razer_bcd_controller -p`. I think I know what it's doing, basically the daemon is only really designed for the chroma keyboard

DasHammett, 2016-06-05:
Hi, 

Thanks for the reply. When I pause the rendering, the blinking stops, but the brightness also gets down by a lot.

Will there be a fix for this? Or you will just concentrate on the chroma keyboards?

Thank you

Kind regards,

terrycain, 2016-06-05:
@DasHammett Yeah basically the daemon is trying to control your keyboard like its a chroma by feeding it RGB values (it accepts RGB but only uses the G part). Technically the brightness is the same but I believe the shade of green has been lowered. 

There will not be a fix for this, I'm working on a completely new daemon written in python which handles multi-device support better which will solve this issue but it's not ready yet (nearly but not quite). I'm moving house soon (like literally in the next few weeks) and starting a new job so will be inactive in terms of development for a while, Ideally I want my new daemon to be live by end of July but were probably looking at end of August.

I'll close the issue for now (as I hate open issues :wink:) but a nice (really cool) improvement is on its way. (It can even do ripple effects and turns off the keyboard when the screensaver is on :smile:).

closed 2016-06-05 by terrycain