pez2001/razer_chroma_drivers #136: Razer Blade Stealth: Not Working

michael-lash, 2016-06-06:
Hello,

After struggling with this for awhile (i.e., found I needed to install the appropriate kernel headers), I believe I have things installed.  Although, when running the ./package_for_debian.sh there was an issue with copying the tmp.*.deb file so I just ran that in place of, what I assume to be, the renamed and copied version (if not, maybe that's the reason I'm having issues).

At any rate, when I run any of the files in the scripts/driver subdirectory I get: "Permission denied"

Additionally, when I run the get_device_type.sh file I get: "No such file or directory", so the reason for my issues may be that it just can't detect the keybaord.

I have also tried razer_bcd_controller -C 8 "bars example" "new rendering node", which results in: 

process 20369: arguments to dbus_message_iter_append_basic() were incorrect, assertion "_dbus_check_is_valid_utf8 (*string_p)" failed in file ../../dbus/dbus-message.c line 2712.
This is normally a bug in some application using the D-Bus library.
daemon controller error: Out of memory!

As an added note I'm using Ubuntu 15.10 and have tried to install the Polychromatic gui which won't start for some reason and have also tried running modprobe razerkbd.

Thanks for any help you can offer. Really looking forward to getting this working!

terrycain, 2016-06-06:
Am moving around a bit, will try and resolve this once I've moved house. 

Does `sudo modprobe razerkbd` work?

Can you bind the device to the keyboard driver? See [this](https://github.com/pez2001/razer_chroma_drivers/wiki/Setting-up-the-keyboard-driver) page for more details

lah7, 2016-06-06:
It sounds like the service isn't running. Try this:

    sudo service razer_bcd restart

 If I recall, at one point, `./package_for_debian.sh` doesn't start the service on Ubuntu 16.04, but `./package_for_ubuntu.sh` does.

> As an added note I'm using Ubuntu 15.10 and have tried to install the Polychromatic gui which won't start for some reason

Polychromatic won't function if the service isn't running. #137 contains a tweak to the driver's modules so that the application can show a graphical error message when the service isn't present.

michael-lash, 2016-06-06:
Thanks both of you for your prompt feedback and suggestions!

Prior to seeing these and after I posted here, I realized I had not yet done a restart of my machine, so I did that.  After that I went back to testing the scripts, but instead of those in the "driver" subdirectory, I went to the "test subdirectory".  I ran the "activate_driver.sh" and "bind_driver.sh" scripts before running "test_daemon.sh" -- I got several of the

process 20369: arguments to dbus_message_iter_append_basic() were incorrect, assertion "_dbus_check_is_valid_utf8 (*string_p)" failed in file ../../dbus/dbus-message.c line 2712.
This is normally a bug in some application using the D-Bus library.
daemon controller error: Out of memory!

errors, but low-and-behold my keyboard colors ended up changing.  After this I test some of the over scripts, but nothing happened.  So, I went to the "driver" subdirectory and retried those.  The result was that my keys were no longer colored (so something happened).  After this, I ran the Polychromatic gui and it started!  I then was able to change the coloring of my keys!  So things seem to be working, just not using the scripts provided -- I'm hesitant to try them again now that things appear to be working....

Also, and this is something perhaps better addressed to the Polychromatic github page, when I tried a "custom" key-coloring scheme, some of the keys failed to light up even though I had marked them.  These included the "Z" key, the "left arrow key", the "fn" and "ctrl" keys on the right side and the "F1" key.

At any rate, things seem to be working just using one of the predefined schemes, so I'm happy!  Thanks again for getting back to me and for this project!

terrycain, 2016-06-06:
What physical keyboard layout are you using? (that might be attributed to the layout possibly)

michael-lash, 2016-06-06:
I'm using the Razer Blade Stealth keyboard

lah7, 2016-06-07:
> some of the keys failed to light up even though I had marked them. These included the "Z" key, the "left arrow key", the "fn" and "ctrl" keys on the right side and the "F1" key.

Ah, that has already been reported here: https://github.com/lah7/polychromatic/issues/11 - I'm not sure why though. :confused:

terrycain, 2016-06-07:
Well it seems to be working so that's this issue done.

closed 2016-06-07 by terrycain