pez2001/razer_chroma_drivers #138: Install does not work and uninstall fails. [ARCH]

Toqozz, 2016-06-12:
Hi, I'm trying to install on arch.

Following the issue reported for compiling on arch, I used:

```
sudo make fedora_install
```

This seems to execute successfully.  (I can then start the daemon `systemctl start razer_bcd`, and lsmod shows that razerkbd is loaded), however trying to use polymetric-controller fails:

```
Traceback (most recent call last):
  File "/usr/bin/polychromatic-controller", line 30, in <module>
    import razer.daemon_dbus
ImportError: No module named 'razer'
```

Also, most commands (such as `razer_bcd_controller -3`) fail: `fish: “razer_bcd_controller -3” terminated by signal SIGSEGV (Address boundary error)`. The error is much more broad in bash.

So, I thought that I should do a clean reinstall -- no luck:
```
 [0][~/g/razer_chroma_drivers] ————— sudo make uninstall
\n::\033[34m Uninstalling Razer kernel modules\033[0m
=====================================================
cp: missing destination file operand after '///lib/modules/4.5.4-1-ARCH/kernel/drivers/usb/misc/razerkbd.ko'
Try 'cp --help' for more information.
Makefile:43: recipe for target 'driver_uninstall' failed
make: *** [driver_uninstall] Error 1
```

Could you tell me the files/folders to manually remove or how to solve these issues?

Thank you for any help!

terrycain, 2016-06-16:
Well if the daemon starts I think it sets things to green so it's a start. Look at where it installed the python library, it probably installed it into the wrong place for arch

Toqozz, 2016-06-20:
So I had to move `/usr/lib/python3/dist-packages/polychromatic` and `/usr/lib/python3/dist-packages/polychromatic` to `/usr/lib/python3.5/site-packages/` (dist-packages would probably work too but I didn't have a directory for it already.

Polychromatic can now start, but gives errors about not being able to connect to the razer_bcd service, which after checking, is indeed not started.

`systemctl status razer_bcd` gives the following output:

```
● razer_bcd.service - Razer Keyboard Daemon
   Loaded: loaded (/usr/lib/systemd/system/razer_bcd.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Mon 2016-06-20 11:06:45 AEST; 8min ago
  Process: 287 ExecStopPost=/usr/share/razer_bcd/systemd_helpers.sh unbind (code=exited, status=0/SUCCESS)
  Process: 278 ExecStart=/usr/sbin/razer_bcd --pid-file /var/run/razer_bcd.pid (code=exited, status=0/SUCCESS)
  Process: 253 ExecStartPre=/usr/share/razer_bcd/systemd_helpers.sh bind (code=exited, status=0/SUCCESS)
 Main PID: 282 (code=exited, status=1/FAILURE)

Jun 20 11:06:45 blade systemd[1]: Starting Razer Keyboard Daemon...
Jun 20 11:06:45 blade razer_bcd[278]: Starting razer blackwidow chroma daemon as a daemon
Jun 20 11:06:45 blade systemd[1]: razer_bcd.service: PID file /var/run/razer_bcd.pid not readable (yet?) after start: No such file or direc
Jun 20 11:06:45 blade systemd[1]: Started Razer Keyboard Daemon.
Jun 20 11:06:45 blade systemd[1]: razer_bcd.service: Main process exited, code=exited, status=1/FAILURE
Jun 20 11:06:45 blade systemd_helpers.sh[287]: ls: cannot access '/sys/bus/hid/drivers/razerkbd/': No such file or directory
Jun 20 11:06:45 blade systemd_helpers.sh[287]: ls: cannot access '/sys/bus/hid/drivers/razermouse/': No such file or directory
Jun 20 11:06:45 blade systemd_helpers.sh[287]: ls: cannot access '/sys/bus/hid/drivers/razerfirefly/': No such file or directory
Jun 20 11:06:45 blade systemd[1]: razer_bcd.service: Unit entered failed state.
Jun 20 11:06:45 blade systemd[1]: razer_bcd.service: Failed with result 'exit-code'.
```

Googling the problem finds #56 but it is closed without any real solution.

`modprobe razerkbd` results in `modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.6.2-1-ARCH`

Toqozz, 2016-06-20:
Alright, just needed to recompile. (`sudo make fedora_install` for Arch)

I didn't have to move any files because they were already moved (there are a lot of junk files around though).

I would suspect that maybe my python path is screwed up, it should have included 3.4 and 3 modules normally I would think.

Thanks for all your help!

closed 2016-06-20 by Toqozz

terrycain, 2016-06-20:
Nah your python path is correct, it's just different to what other distributions use

lah7, 2016-06-20:
Thanks for the info @Toqozz, I'll update Polychromatic's install script to look up the path in future.

One question though, what is your output for this?

    python3 -c "import sys; print(sys.path[-1])"

Toqozz, 2016-06-20:
```
 [0][~] ————— python3 -c "import sys; print(sys.path[-1])"
/usr/lib/python3.5/site-packages
```