pez2001/razer_chroma_drivers #141: Blackwidow x chroma support?

geordieF, 2016-06-16:
it is 1532:0216

terrycain, 2016-06-16:
List all of the features it has, ideally screenshots from Synapse will help. Also am moving house so might not be able to do much for a while.

geordieF, 2016-06-16:
Same as blackwidow chroma except with no macro keys

terrycain, 2016-06-17:
Ok I'll make up a driver for you to try out later

terrycain, 2016-06-17:
Read [this](https://github.com/pez2001/razer_chroma_drivers/wiki/Setting-up-the-keyboard-driver) page and become familiar with binding the device to a driver, as that will need to be done to test.

Then, read [this](https://github.com/pez2001/razer_chroma_drivers/issues/89#issuecomment-221058019) issue's comments from that comment onwards.

On my fork of the razer repository there is a [branch](https://github.com/terrycain/razer_drivers/tree/RazerBlackWidowXChroma_0x0216) called `RazerBlackWidowXChroma_0x0216` which you should use instead of the one on that issue thread.

Got it?