pez2001/razer_chroma_drivers #143: Mouse functionality lost

01generator, 2016-07-02:
Hello,

First of i have a reazer chroma x blackwidow keyboard and a razer abyssus mouse.

I had installed ubuntu 16.04LTS and after razer_chroma_drivers installation mouse functionality got lost. 
I have made a clean install of ubuntu 16.04LTS and installed the razer_chroma_drivers again, result was the same.
After that I downloaded kubuntu 14.04LTS and installed it, result was the same. 

A wild guess is that 2 razer products and specifically abyssus mouse is causing trouble. 

Any fixes on these? I really want to use my keyboard features, I do not care about the mouse, if somehow I can isolate it to be functioning just for the keyboard would be great. 

Keep up the good work!

terrycain, 2016-07-02:
Plug in an additional mouse so you can still click. Install the drivers (so that your abyssus doesn't work) then run `ls -R /sys/bus/hid/drivers`

In theory both should work fine together as there are technically separate drivers. Most likely the wrong mouse USB interface is getting rebound to the razermouse driver.

01generator, 2016-07-05:
Ok I have installed the drivers again.

Usually problem starts after a reboot.

This is the outcome of ls -R /sys/bus/hid/drivers before reboot

/sys/bus/hid/drivers:
hid-generic

/sys/bus/hid/drivers/hid-generic:
0003:045E:0040.0005  0003:1532:0216.0002  0003:1532:0216.0004  module  uevent
0003:1532:0042.0001  0003:1532:0216.0003  bind                 new_id  unbind

01generator, 2016-07-05:
Correction! 

After reboot razer mouse only works, the other mouse lost functionality

terrycain, 2016-07-05:
The other mouse doesn't work at all? Thats most odd, did the keyboard work?

01generator, 2016-07-06:
Ok correction cause I am a fool, both mice working properly I removed it to see if the abyssus was working solo and I forgot it. 

Now I have a clean install of kubuntu and razer chroma drivers install with both mice pluged in and working 
With output of ls -R /sys/bus/hid/drivers 

/sys/bus/hid/drivers:
hid-generic  razermouse

/sys/bus/hid/drivers/hid-generic:
0003:045E:0040.0001  0003:1532:0216.0004  bind    new_id  unbind
0003:1532:0216.0003  0003:1532:0216.0005  module  uevent

/sys/bus/hid/drivers/razermouse:
0003:1532:0042.0002  bind  module  new_id  uevent  unbind

Now I am gone install polychromic and come back to you.

terrycain, 2016-07-06:
Well it seems the razerkbd driver isnt being loaded :worried:. + polychromic will not work with abyssus as the daemon doesnt support it I dont think

01generator, 2016-07-06:
So as you said polychromic is not working at all, I do not know if razerkbd is working hmmm

But we managed to overpass the no working mouse.

If you install the driver with 2 mouse connected, reboot, remove the 2nd mouse and reboot again all is working as expected. 

Any code for terminal to test if keyboard is accepting commands?

terrycain, 2016-07-06:
Well this tells us its not mounted as there should be a `razerkbd` entry.
```
/sys/bus/hid/drivers:
hid-generic razermouse
```

try `sudo modprobe razerkbd` and if that works run `ls -R /sys/bus/hid/drivers`
Also format the output of the terminal using (remove the spaces between the backticks though)
```
` ` `
Test
` ` `
```

01generator, 2016-07-06:
So the output after the 2 commands

```
hid-generic  razerkbd  razermouse

/sys/bus/hid/drivers/hid-generic:
0003:1532:0216.0002  0003:1532:0216.0004  module  uevent
0003:1532:0216.0003  bind                 new_id  unbind

/sys/bus/hid/drivers/razerkbd:
bind  module  new_id  uevent  unbind

/sys/bus/hid/drivers/razermouse:
0003:1532:0042.0001  bind  module  new_id  uevent  unbind
```

Note that second mouse is not plugged in.

terrycain, 2016-07-06:
Thats fine. Ok the keyboard is not binded to the keyboard driver but at least the keyboard driver is loaded. Try `sudo udevadm control --reload-rules` then rerun the ls -R

01generator, 2016-07-06:
Hmmm I do not think that the outcome is different

```
/sys/bus/hid/drivers:
hid-generic  razerkbd  razermouse

/sys/bus/hid/drivers/hid-generic:
0003:1532:0216.0002  0003:1532:0216.0004  module  uevent
0003:1532:0216.0003  bind                 new_id  unbind

/sys/bus/hid/drivers/razerkbd:
bind  module  new_id  uevent  unbind

/sys/bus/hid/drivers/razermouse:
0003:1532:0042.0001  bind  module  new_id  uevent  unbind
```

terrycain, 2016-07-06:
It's not :-(, am on the train. I'll give you a command later to try and bind the driver manually

01generator, 2016-07-11:
Hello,
Any news updates on the subject?

terrycain, 2016-07-11:
Sorry. Try this (either run both those commands in a script or use a 2nd keyboard as after the first 1 the keyboard will be unresponsive until running the 2nd)
```
echo -n '0003:1532:0216.0004' > /sys/bus/hid/drivers/hid-generic/unbind
echo -n '0003:1532:0216.0004' > /sys/bus/hid/drivers/razerkbd/bind
```

Then run `ls -R /sys/bus/hid/drivers` again. In theory the keyboard should be bound to the razer keyboard driver then

01generator, 2016-07-11:
ls -R /sys/bus/hid/drivers before the commands you listed

```
/sys/bus/hid/drivers:
hid-generic  razermouse

/sys/bus/hid/drivers/hid-generic:
0003:1532:0216.0001  bind    uevent
0003:1532:0216.0002  module  unbind
0003:1532:0216.0003  new_id

/sys/bus/hid/drivers/razermouse:
0003:1532:0042.0004  module  uevent
bind                 new_id  unbind

```

And result in the command you listed 

`sudo echo -n '0003:1532:0216.0004' > /sys/bus/hid/drivers/hid-generic/unbind`
`bash: /sys/bus/hid/drivers/hid-generic/unbind: Permission denied`

I also tried `sudo echo -n '0003:1532:0216.0003' > /sys/bus/hid/drivers/hid-generic/unbind`
Result was the same.

claell, 2016-07-11:
I think you need to be in root shell to do this, but I may be wrong.

01generator, 2016-07-11:
You are correct claell

I did `echo -n '0003:1532:0216.0003' > /sys/bus/hid/drivers/hid-generic/unbind`
No outoput - all ok i guess

Then `echo -n '0003:1532:0216.0003' > /sys/bus/hid/drivers/razerkbd/bind`
Output `bash: /sys/bus/hid/drivers/razerkbd/bind: No such file or directory`
Hmmmm O.o 

`ls -R /sys/bus/hid/drivers`

```
/sys/bus/hid/drivers:
hid-generic  razermouse

/sys/bus/hid/drivers/hid-generic:
0003:1532:0216.0001  bind    new_id  unbind
0003:1532:0216.0002  module  uevent

/sys/bus/hid/drivers/razermouse:
0003:1532:0042.0004  module  uevent
bind                 new_id  unbind
```

terrycain, 2016-07-11:
razerkbd driver seems to have unloaded itself. try `modprobe razerkbd` then `echo -n '0003:1532:0216.0003' > /sys/bus/hid/drivers/razerkbd/bind`

d-rez, 2016-08-09:
I had a similar issue where the mouse stopped working right after installing the driver

```
Aug 09 22:21:04 razer_bcd[9527]: Binding device (0003:1532:0045.0007) to razermouse
Aug 09 22:21:04  razer_bcd[9527]: /etc/init.d/razer_bcd: 22: /etc/init.d/razer_bcd: cannot create /sys/bus/hid/drivers/razer
Aug 09 22:21:04  systemd[1]: razer_bcd.service: Control process exited, code=exited status=2
Aug 09 22:21:04  systemd[1]: Failed to start LSB: start/stops razer_bcd.

```
Managed to fix this by manually enabling the kernel module
`$ modprobe razermouse`
and restarting the service
`$ systemctl status razer_bcd.service`