pez2001/razer_chroma_drivers #145: `Configured resource limit exceeded`??

geordieF, 2016-07-06:
When I try to start the service with `sercive razer_bcd start` I get 
> Job for razer_bcd.service failed because a configured resource limit was exceeded. See "systemctl status razer_bcd.service" and "journalctl -xe" for details.

terrycain, 2016-07-06:
Not a clue :wink:, not running out of ram are you?

geordieF, 2016-07-07:
12gb of it? nah.

terrycain, 2016-07-07:
Nah each process technically has a limit assigned to it. It could also be the daemon accessing parts of ram it's not supposed to.

Am waiting on a few things before I set up my pc again but once it's up you can be one of the first testers of the new daemon :wink:

JudeSeide, 2016-07-10:
I'm facing a similar issue trying to install this driver on my new blade 14 2016 Ubuntu 16.04, any progress on the new daemon?

terrycain, 2016-07-10:
Good news, I've now set up my PC :smile:, Just wiped my PC so putting some stuff back on it. 
The new daemon is still in like before alpha, and is experimental and will die alot but if you willing to test go to my fork and post in [this](https://github.com/terrycain/razer_drivers/issues/5) issue, and that will keep you updated, etc...