pez2001/razer_chroma_drivers #146: Raspbian ?

xelaco, 2016-07-08:
Hello,
I've tried to install the drivers on my Raspberry pi and, obviously, it hasn't worked...
So, I wonder if you will program a version for Raspbian. But, I think it's very hard because the architecture is : armhf.
If you want the errors I have them in... French. You can read [this post](https://www.raspberrypi.org/forums/viewtopic.php?f=65&t=152760&p=1001123) in the official forum in French.

terrycain, 2016-07-09:
Ok basically I've moved house and am waiting to set up my pc. On said pc I have a replacement for the currentl daemon which is written in python so that there's one less thing to compile.

Does `make driver_verbose` work? Paste the output.

xelaco, 2016-07-10:
Thanks for your very fast answer.
Unfortunately, I'm in holidays and I left my Raspberry pi at home. I can't recover it before one month and a half... So I can't try anything.
I would post a comment here once I have recovered it or if I find one on the floor. :)

terrycain, 2016-07-10:
No problem

d-rez, 2016-08-09:
Hey there,

I installed prerequisites from scripts + raspberrypi-kernel-headers, 

Here's what I'm getting:

`pi@raspberrypi:~/src/razer_chroma_drivers $ make
make --no-print-directory -C lib all
gcc -Wall -Wextra -W -w -O0 -std=c99 -DMAJOR_VERSION=0 -DMINOR_VERSION=3 -fPIC -DPIC -DLINUX -D_XOPEN_SOURCE=700 `pkg-config --cflags dbus-1` -DUSE_DBUS -c -o razer_chroma.o razer_chroma.c
gcc -Wall -Wextra -W -w -O0 -std=c99 -DMAJOR_VERSION=0 -DMINOR_VERSION=3 -fPIC -DPIC -DLINUX -D_XOPEN_SOURCE=700 `pkg-config --cflags dbus-1` -DUSE_DBUS -c -o list.o list.c
gcc -Wall -Wextra -W -w -O0 -std=c99 -DMAJOR_VERSION=0 -DMINOR_VERSION=3 -fPIC -DPIC -DLINUX -D_XOPEN_SOURCE=700 `pkg-config --cflags dbus-1` -DUSE_DBUS -c -o razer_string.o razer_string.c
ar -c -rs librazer_chroma.a razer_chroma.o list.o razer_string.o
gcc -rdynamic -nostdlib -DCREATELIB -shared razer_chroma.o list.o razer_string.o -lm -lc `pkg-config --libs dbus-1` -o librazer_chroma.so
gcc -Wall -Wextra -W -w -O0 -std=c99 -DMAJOR_VERSION=0 -DMINOR_VERSION=3 -fPIC -DPIC -DLINUX -D_XOPEN_SOURCE=700 `pkg-config --cflags dbus-1` -DUSE_DBUS -c -o razer_chroma_controller.o razer_chroma_controller.c
gcc -rdynamic -nostdlib -DCREATELIB -shared razer_chroma_controller.o list.o razer_string.o -lm -lc `pkg-config --libs dbus-1` -o librazer_chroma_controller.so
ar -c -rs librazer_chroma_controller.a razer_chroma_controller.o list.o razer_string.o

:: COMPILING librazer_chroma.so
================================
gcc -Wall -Wextra -Wformat=0 -pedantic -fbounds-check -Wuninitialized -O0 -g3 -std=c99 -DUSE_DEBUGGING -DMAJOR_VERSION=0 -DMINOR_VERSION=3 -fPIC -DPIC -DLINUX -D_XOPEN_SOURCE=700 `pkg-config --cflags dbus-1` -DUSE_DBUS -c -o razer_chroma.do razer_chroma.c
razer_chroma.c: In function ‘razer_find_devices’:
razer_chroma.c:643:19: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
     event.value = (unsigned long long)device;
                   ^
gcc -Wall -Wextra -Wformat=0 -pedantic -fbounds-check -Wuninitialized -O0 -g3 -std=c99 -DUSE_DEBUGGING -DMAJOR_VERSION=0 -DMINOR_VERSION=3 -fPIC -DPIC -DLINUX -D_XOPEN_SOURCE=700 `pkg-config --cflags dbus-1` -DUSE_DBUS -c -o list.do list.c
gcc -Wall -Wextra -Wformat=0 -pedantic -fbounds-check -Wuninitialized -O0 -g3 -std=c99 -DUSE_DEBUGGING -DMAJOR_VERSION=0 -DMINOR_VERSION=3 -fPIC -DPIC -DLINUX -D_XOPEN_SOURCE=700 `pkg-config --cflags dbus-1` -DUSE_DBUS -c -o razer_string.do razer_string.c
ar -c -rs librazer_chroma.da razer_chroma.do list.do razer_string.do
gcc -rdynamic -nostdlib -DCREATELIB -shared razer_chroma.do list.do razer_string.do -lm -lc `pkg-config --libs dbus-1` -o librazer_chroma_debug.so
gcc -Wall -Wextra -Wformat=0 -pedantic -fbounds-check -Wuninitialized -O0 -g3 -std=c99 -DUSE_DEBUGGING -DMAJOR_VERSION=0 -DMINOR_VERSION=3 -fPIC -DPIC -DLINUX -D_XOPEN_SOURCE=700 `pkg-config --cflags dbus-1` -DUSE_DBUS -c -o razer_chroma_controller.do razer_chroma_controller.c
razer_chroma_controller.c: In function ‘dc_render_node_parameter_parsed_set’:
razer_chroma_controller.c:345:10: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
    value=(unsigned long long)value_string;
          ^
razer_chroma_controller.c:358:13: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
     value = (unsigned long long)malloc(sizeof(struct razer_rgb));
             ^
razer_chroma_controller.c:366:6: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
     ((struct razer_rgb*)value)->r = (unsigned char)atoi(list_Dequeue(tokens));
      ^
razer_chroma_controller.c:367:6: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
     ((struct razer_rgb*)value)->g = (unsigned char)atoi(list_Dequeue(tokens));
      ^
razer_chroma_controller.c:368:6: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
     ((struct razer_rgb*)value)->b = (unsigned char)atoi(list_Dequeue(tokens));
      ^
razer_chroma_controller.c:376:13: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
     value = (unsigned long long)malloc(sizeof(struct razer_pos));
             ^
razer_chroma_controller.c:384:6: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
     ((struct razer_pos*)value)->x = (long)atol(list_Dequeue(tokens));
      ^
razer_chroma_controller.c:385:6: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
     ((struct razer_pos*)value)->y = (long)atol(list_Dequeue(tokens));
      ^
razer_chroma_controller.c:396:13: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
     value = (unsigned long long)malloc(sizeof(struct razer_float_range));
             ^
razer_chroma_controller.c:404:6: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
     ((struct razer_float_range*)value)->min = (float)atof(list_Dequeue(tokens));
      ^
razer_chroma_controller.c:405:6: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
     ((struct razer_float_range*)value)->max = (float)atof(list_Dequeue(tokens));
      ^
razer_chroma_controller.c:413:13: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
     value = (unsigned long long)malloc(sizeof(struct razer_int_range));
             ^
razer_chroma_controller.c:421:6: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
[... cut ...]
make[1]: *** [daemon_shared] Error 1
Makefile:59: recipe for target 'daemon' failed
make: *** [daemon] Error 2
`

`pi@raspberrypi:~/src/razer_chroma_drivers $ make driver_verbose

:: Compiling Razer kernel modules
========================================
make -C /lib/modules/4.4.16-v7+/build SUBDIRS=/home/pi/src/razer_chroma_drivers/driver modules
make[1]: *** /lib/modules/4.4.16-v7+/build: No such file or directory.  Stop.
Makefile:24: recipe for target 'driver_verbose' failed
make: *** [driver_verbose] Error 2
`

terrycain, 2016-08-09:
Mind recreating this issue on my [fork](https://github.com/terrycain/razer_drivers) with just the output of `make driver_verbose` and also the output of `ls -lah /lib/modules/`

d-rez, 2016-08-10:
@terrycain sure thing, will do it this evening. As it's a rPi I'm aware not all modules might be installed but I couldn't find a package responsible for this particular one. Do you want me to create the issue on your fork and close this one?

terrycain, 2016-08-10:
Yeah, it's raspbian so in theory it should have the kernel headers somewhere

xelaco, 2016-08-22:
I came back to my home and my Raspberry pi is near to me. (Next year, I will check more attentively my bag...)
What do you want me I try ? (I'm not sure about the syntax of this sentence...)

terrycain, 2016-08-22:
First ensure the kernel headers are installed.

xelaco, 2016-08-26:
I've  a little problem... I can't install the kernel headers.... (Raspbian didn't start when I changed the boot config to install the kernel headers)  I've still some difficulties with Linux ! Could you help me ?

terrycain, 2016-08-26:
You should be able to do `sudo apt-get install linux-headers-$(uname -r)` that should install them for you

xelaco, 2016-08-26:
Unfortunately, the package doesn't exist... apt-get can't find it.

terrycain, 2016-08-26:
What does this return `sudo apt-cache search linux-header*`

d-rez, 2016-08-26:
Huh... That's interesting

```
pi@raspberrypi:~ $ sudo apt-cache search linux-header*
linux-headers-3.10-3-all - All header files for Linux 3.10 (meta-package)
linux-headers-3.10-3-all-armhf - All header files for Linux 3.10 (meta-package)
linux-headers-3.10-3-common - Common header files for Linux 3.10-3
linux-headers-3.10-3-rpi - Header files for Linux 3.10-3-rpi
linux-headers-3.12-1-all - All header files for Linux 3.12 (meta-package)
linux-headers-3.12-1-all-armhf - All header files for Linux 3.12 (meta-package)
linux-headers-3.12-1-common - Common header files for Linux 3.12-1
linux-headers-3.12-1-rpi - Header files for Linux 3.12-1-rpi
linux-headers-3.16.0-4-all - All header files for Linux 3.16 (meta-package)
linux-headers-3.16.0-4-all-armhf - All header files for Linux 3.16 (meta-package)
linux-headers-3.16.0-4-common - Common header files for Linux 3.16.0-4
linux-headers-3.16.0-4-rpi - Header files for Linux 3.16.0-4-rpi
linux-headers-3.18.0-trunk-all - All header files for Linux 3.18 (meta-package)
linux-headers-3.18.0-trunk-all-armhf - All header files for Linux 3.18 (meta-package)
linux-headers-3.18.0-trunk-common - Common header files for Linux 3.18.0-trunk
linux-headers-3.18.0-trunk-rpi - Header files for Linux 3.18.0-trunk-rpi
linux-headers-3.18.0-trunk-rpi2 - Header files for Linux 3.18.0-trunk-rpi2
linux-headers-3.6-trunk-all - All header files for Linux 3.6 (meta-package)
linux-headers-3.6-trunk-all-armhf - All header files for Linux 3.6 (meta-package)
linux-headers-3.6-trunk-common - Common header files for Linux 3.6-trunk
linux-headers-3.6-trunk-rpi - Header files for Linux 3.6-trunk-rpi
linux-headers-4.4.0-1-all - All header files for Linux 4.4 (meta-package)
linux-headers-4.4.0-1-all-armhf - All header files for Linux 4.4 (meta-package)
linux-headers-4.4.0-1-common - Common header files for Linux 4.4.0-1
linux-headers-4.4.0-1-rpi - Header files for Linux 4.4.0-1-rpi
linux-headers-4.4.0-1-rpi2 - Header files for Linux 4.4.0-1-rpi2
linux-headers-rpi - Header files for Linux rpi configuration (meta-package)
linux-headers-rpi-rpfv - This metapackage will pull in the headers for the raspbian kernel for the
linux-headers-rpi2-rpfv - This metapackage will pull in the headers for the raspbian kernel for the
raspberrypi-kernel-headers - Header files for the Raspberry Pi Linux kernel
```

```
pi@raspberrypi:~ $ sudo apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-headers-4.4.17-v7
E: Couldn't find any package by regex 'linux-headers-4.4.17-v7'

```

terrycain, 2016-08-26:
Are you running a custom kernel?

xelaco, 2016-08-30:
No.
I think that kernel headers aren't included in Raspbian, but I found that : https://www.raspberrypi.org/forums/viewtopic.php?f=71&t=17666&p=179845
I will try this evening.

xelaco, 2016-09-01:
The Raspberry Foundation doesn't let us access to kernel headers easily...
I tried the link and it doesn't work... I think that I found the solution using rpi-source but there is an error with make. I think it's impossible that Razer driver work on Raspbian.

GabMus, 2016-09-01:
Maybe you could just try a more friendly distro on your raspberry pi. Maybe
arch?