pez2001/razer_chroma_drivers #147: [Need testers/enhancement] help testing gtk app for razer_chroma_drivers

GabMus, 2016-07-10:
Hello there, I just wanted to share with you a frontend I'm developing for this driver.

It's called razerCommander, it's licensed under GPL 3, you can find it here:
https://github.com/gabmus/razerCommander

I'd need testers, since the only device I could test this with is my blackwidow ultimate 2016, so no rgb settings couldn't be tested. Thank you.

terrycain, 2016-07-10:
ooohhhh @lah7 you got yourself some competition now :wink: hehe. @GabMus I'm working on updating the daemon, once that's become a bit more stable i'll be happy to try it out.

GabMus, 2016-07-10:
> now 😉 hehe. @GabMus <https://github.com/GabMus> I'm working on updating

That's great! To be honest this app doesn't use the daemon as I couldn't
make it work, I just write on the files in /sys/bus/... manually

terrycain, 2016-07-10:
> doesn't use the daemon as I couldn't
> make it work

Your not alone ;), the old daemon doesn't handle multiple devices well. The new daemon will be nicer as eventually there will be a python library you can just import and that'll do all the heavy lifting for you.

GabMus, 2016-07-10:
That's great! I made my little dirty library (the file called device.py), and for now it will do, until the daemon is ready. When it is I just have to edit that file and I'm done! :smile:

rivetrik, 2016-07-28:
i am willing to test this out tonight when i get home on my blade stealth, looks promising!

rivetrik, 2016-07-29:
ok , so i did some testing on it and heres what happend, first of all, it did work , pretty well except some of the light profiles didnt register (wave and spectrum), second there was no ripple effect like the picture showed, not sure if that would be a bug or not, but i thought i would mention it and lastly, when i bound the chroma driver to my RBS keyboard, it made the application non functional. it will still open but changing profiles is a no go

GabMus, 2016-07-29:
Yeah, well. There's a bit of a transition going on right now. First, I'm
currently working with @terrycain 's fork of this driver, which is more up
to date. Second, there are some little issue with the current version of
razerCommander, but you can easily fix them by switching to the
experimental customSupport branch. About the starlight effect, that needs
a  hardware implementation, and apparently your keyboard doesn't support
it. Maybe in the future it will be possible to implement it via software,
just like the ripple effect, but it's too early to speculate on this.
Finally there's currently some work going on for a python library that
communicates to the dbus daemon, so there will be a lot of changes (in
better!)

rivetrik, 2016-07-29:
sweet!  cant wait to try it out!!

terrycain, 2016-07-29:
As far as im aware starlight is only available on the blackwidow 2016 keyboards but I may be wrong.

rivetrik, 2016-07-29:
it was one of the stock presets in synapse before i wiped windows off of my stealth, if you guys need a guinea pig for testing let me know, im down for it, anything to help out the driver and app out!

terrycain, 2016-07-29:
Yeah synapse has a habit of showing presets that are actually custom effects. If you want go the wiki, read the section about reverse engineering and get me a packet capture for the star light effect and I'll happily enable it for the stealth. (I could enable it now but I don't like editing the driver for others without some sort of proof it'll work)

rivetrik, 2016-07-29:
i can see what i can do this weekend when i can sit down and focus on getting those packets for you, i mean in all honesty its not that big of a deal to have that particular effect, i'd be more happy having the ability to use custom lighting animations and what not, but i understand that probably wont be the case for a while. but i want to help out any way i can

terrycain, 2016-07-30:
Yeah I know what you mean. No rush, am currently away from pc for a while so only have my laptop to do stuff on