pez2001/razer_chroma_drivers #148: Kraken Chroma support

z3ntu, 2016-07-11:
Changing the colors please :)

terrycain, 2016-07-11:
Read [this](https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol), (but obviosuly substitute keyboard for headphones). If you can manage that and are willing to help then sure.

z3ntu, 2016-07-11:
Thanks. I will try to add support.
Is there a page where all steps to add a new device are listed? Like "add a new line in this file" "and in that file" and "then implement support in this file"?

terrycain, 2016-07-11:
Nah its simpler than that. You capture some usb packets with wireshark and send them to me. I'll make a driver, send it to you to test then jobs done.

z3ntu, 2016-07-11:
Okay, that's also possible ;) I can send them to you tomorrow evening (GMT+2)

terrycain, 2016-07-11:
No problem. Send them whenever is good for you. Also could I get the output of `lsusb`

z3ntu, 2016-07-12:
I have some old (January) capture files with some notes here, maybe they help. If not, I can capture more, I just don't have the time right now (doing a holiday internship where the travel time alone is about 2 hours a day and I have to work about 8 hours).
[Kraken.tar.gz](https://github.com/pez2001/razer_chroma_drivers/files/360200/Kraken.tar.gz)

terrycain, 2016-07-12:
Oooooh looks like were going to have some fun here. I have the spec of the protocol format and it doesn't match what I have here so they have used a different protocol/format. Its useful but not enough for me to comfortably make a driver.

Firstly I need the output of `lsusb -d 1532:0504 -vvv`. I also need a print screen of all the options on the razer synapse windows GUI if you can get to it (that would help immensely). Then when you've got some time say so and ill post a whole list of things to do whilst capturing packets (so we can isolate what packet does what in more detail as there is packets that do things other than colour, possibly volume?).

Also if your up for it at some point I'd like to get some captures from your deathadder too, to add to the collection.

z3ntu, 2016-07-13:
About the Deathadder (Chroma): razercfg already has support for it, maybe you can look if that source code can help you.
About the Kraken (Chroma): I'll try to do this in the evening.

z3ntu, 2016-07-13:
The output of the lsusb command: http://pastebin.com/wfUk5FiS
The other things will come.

z3ntu, 2016-07-13:
I hope this helps you now:
[kraken_compressed.zip](https://github.com/pez2001/razer_chroma_drivers/files/362036/kraken_compressed.zip) (GitHub doesn't like .tar.xz) :(

terrycain, 2016-07-13:
Cheers, I'm looking at it now. Have also emailed my contacts at #Razer to see if they want to help out too.

I've decoded the static sequence and what I "think" could be the protocol but the other lighting modes have eluded me. Also mic and volume level is done in software.

Could I get 1 capture with you switching between static and breath like 4 times
1 switching between static and specturm 4 times,
1 switching between static and none 4 times

Also I want to find out if the kraken has serial number capability as we use that in the daemon to differentiate between devices. Could you start the capture before turning on the vm, keep going, open synapse and once that's fully loaded then stop the capture. 

But all in all were making good progress :)

> razercfg already has support for it

It does but its not a kernel driver and not particularly helpful, packet captures would be most helpful ha :wink: Someone else was gonna get me captures but they went afk.

z3ntu, 2016-07-13:
[Kraken.zip](https://github.com/pez2001/razer_chroma_drivers/files/362214/Kraken.zip)
I hope here is everything :)

terrycain, 2016-07-13:
Wow that was quick. Will get on it

terrycain, 2016-07-13:
Ite, familiarise yourself with [this](https://github.com/pez2001/razer_chroma_drivers/wiki/Setting-up-the-keyboard-driver) page as you'll need to manually bind the driver once i've made it until its been tested.

Still not 100% on it as it seems to send 4 packets to get done what 1 would do on the keyboard. But I will just replay to the headset what I've been given :wink:. You forgot static to spectrum but no rush. I'll make a driver next week, currently working on getting my daemon tested :smile:.

z3ntu, 2016-07-21:
Any news? (no need to hurry, just curious)

terrycain, 2016-07-21:
Actaully yes, my Razer contact got back to me this morning (conversations
normally take a while due to the time differences etc...) and they're
working on a kraken and mice spec for me so should be good :)

terrycain, 2016-08-28:
Ugh yeah I haven't forgotten about you :wink:. They are compiling a specification for about 12 devices but they forgot about it the other day. I've prodded them again. Will email again Wednesday.

z3ntu, 2016-09-29:
Hello again ;)

terrycain, 2016-09-29:
Yeah have got a specification of mice, have emailed them for a spec for kraken and the man o war. Slowly getting there

z3ntu, 2016-09-30:
Great to hear! No matter how long it takes as long as everything is working somewhere in the future, I will be happy :)

terrycain, 2016-09-30:
Yeah progress is being made slowly. They're not paid to help so the issue is they have to put time in alongside their job. We'll get there eventually

z3ntu, 2016-12-29:
Closing in favor of https://github.com/terrycain/razer-drivers/issues/57

closed 2016-12-29 by z3ntu