pez2001/razer_chroma_drivers #150: macro keys are disabled after resume from suspend

lukovdm, 2016-07-23:
Hi,

When I resume from suspend the macro keys are disabled, is this a bug or does anyone have a fix?

lah7, 2016-07-23:
Does it work again if you restart the service after resuming from suspend?

    sudo service razer_bcd restart

I experience a _similar_ bug whereby the keyboard cannot be "changed" after resuming from suspend. I **_think_** both issues are due to the service/daemon seeing the keyboard as a "new" device and the daemon has flaky multi-device support at the moment.

Restarting the service (and any applications that interface the daemon) should be a temporary workaround. I have to do this every time after resuming from suspend.

lukovdm, 2016-07-23:
I have polychromatic installed, and i can turn them on again with it after resuming.

terrycain, 2016-07-24:
suspend or hibernate?

m4ng0squ4sh, 2016-07-25:
This should be fixed in the kernel driver and not in the daemon.

@lukovdm Which keyboard do you use?

terrycain, 2016-07-25:
@m4ng0squ4sh Nope, it used to be enabled in the driver long ago but in doing so can disable the Fn+Key combos on most keyboards therefore the decision was made to have macro keys opt in. That's an issue with the current daemon.

m4ng0squ4sh, 2016-07-25:
@terrycain I doubt that it was implemented right. I am just adding support for suspend and resume actions to the [OpenRazer kernel driver](https://github.com/openrazer/razer-drivers). I borrowed a BlackWidow Chroma keyboard from a friend and I'll try to reproduce the described bug.

terrycain, 2016-07-25:
@m4ng0squ4sh No it wasn't implemented right. The right way would to put the keyboard into "driver" mode and write a full keyboard driver. This was done to save writing an entire keyboard driver.

m4ng0squ4sh, 2016-07-25:
@terrycain The right way would be to add the power management kernel support, save the device states as drv data to the kernel device and reset them during a resume. This does not differ from a manual interaction after a resume.

m4ng0squ4sh, 2016-07-26:
I implemented it as described with an automatic macro key activation. Seams to work so far. I'll keep an eye on the FN+Key functionalitity...

https://github.com/openrazer/razer-drivers/commit/2d8681a0086f5d5f26786feba5c5f85c31672e09