pez2001/razer_chroma_drivers #151: Naga Epic Chroma Support

Misterio77, 2016-07-28:
Hey there,
Just got a Razer Naga Epic Chroma, and it seems this driver can't detect it (It's not on the supported list, either).
I was wondering if there was any chance you could try to add support? I would be glad to help in any way i can, too.

Thanks a lot,

terrycain, 2016-07-28:
If its not on the list its most likely its not supported yet. Until my contacts at Razer give me a spec for the mice I can't do anything from here so you'll have to help out. Read [this](https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol) also go through some of the closed tickets for device support (for devices that are now supported) as well.

Once you've read through that link go [here](https://github.com/terrycain/razer_drivers/issues/5#issuecomment-232477960) and read the stuff about Telegram, if you want you can join that as a few of us are active around 1900 UTC. 

First action of business is getting a windows VM, passing the naga through to the VM so razer synapse recognises it, then giving me screenshots of all the options in synapse.

Misterio77, 2016-07-28:
Thanks for the quick reply,
Sure thing, downloading a Windows ISO right away.
My timezone is GMT -3, so i'm not sure if i can be really active in Telegram, but i'll try my best.
Here's a link to my profile: https://telegram.me/Misterio7

terrycain, 2016-07-28:
At work currently. I'll add you when I get home. And np just post stuff on there someone will help you out.

Misterio77, 2016-07-28:
Understood, Thanks a lot.

*|---------------------------------------------------------------------------------------------------------------------|*
*-Misterio7 / Gabriel Fontes*

*| Site Pessoal <http://misterio7.com> | Twitter
<https://twitter.com/Misterio7x7> | Reddit
<https://www.reddit.com/user/Misterio77x/> | YouTube
<https://www.youtube.com/c/Misterio7> |*
*Skype: gsfontes2000 <http://skype:gsfontes2000?chat> | RuneScape: Misterio
<http://services.runescape.com/m=adventurers-log/a=63/profile?searchName=Misterio>
| Minecraft: Misterio7*