pez2001/razer_chroma_drivers #152: Razer Mamba charging effect: speed depends on RGB values

d-rez, 2016-08-09:
Just noticed this funny bug: the smaller values passed to set_charging_colour.py script, the quckier it animates.

This looks normal
`$ sudo python3 set_charging_colour.py --colour 255 255 255`
This looks normal as well
`$ sudo python3 set_charging_colour.py --colour 255 0 0`
Fun starts here, effect animates twice as fast
`$ sudo python3 set_charging_colour.py --colour 127 0 0`
You get the idea. The following makes the effect cycle around 3 times / second
`$ sudo python3 set_charging_colour.py --colour 5 5 5`

terrycain, 2016-08-09:
Have moved all development to my fork, would you mind reraising the issue there and closing this one. Thanks.

d-rez, 2016-08-10:
Yupp will do tonight, cheers!

closed 2016-08-10 by d-rez