pez2001/razer_chroma_drivers #153: Question about models support

proxymus89, 2016-08-23:
Are those hardware support?

- Razor Mamba Wireless model (I've seen that Razor Mamba is supported, but also Wireless model?)
- Razer Man O War
- Razer Blade 14" (2016) Keyboard

If not, will be implemented?

terrycain, 2016-08-23:
Razer Mamba when using the wireless dock is supported. I need to adjust the driver to support it when it's plugged in directly.

Razer Man o War is not supported. But if your willing to help then we could add it.

Razer Blade Stealth one of them is supported so possibly. If it is not then it shouldn't be difficult to add it

proxymus89, 2016-08-23:
Thank you very much :) and yes i would help you to implement new driver (as far as possible), I've already talken with another contributor (gabmus) few hours ago. But actually i've only the blackwidow chroma wich is already support, but in the future i want buy the Mamba,Man o War and Razer blade 14", for this reason I asked :)

m4ng0squ4sh, 2016-08-23:
@proxymus89 Razer Blade 14 2016 is fully supported [here](https://github.com/openrazer/openrazer-drivers).

terrycain, 2016-08-24:
Yeah sure. The more devices we can add the better :-)

terrycain, 2016-08-24:
Feel free to join the telegram chat group we have somewhere

proxymus89, 2016-08-24:
@m4ng0squ4sh Thank you very much, this is a good news for me, I want buy a razer blade pro and I was worried for keyboard keylights support. Do you know by any chance if Razer blade 14 2016 works out of the box with Linux? (i want install Kali, but eventually also Ubuntu), or there are some component wich need some tricks to work?

m4ng0squ4sh, 2016-08-24:
@proxymus89 I had a lot of troubles but got everything fixed now. System-Boot freeze, Kernel problems, GPU stuff, [DSDT firmware fix](https://github.com/m4ng0squ4sh/razer_blade_14_2016_acpi_dsdt), LED lighting modes,...
I really love the Razer Blade 14 and it's a masterpiece of hardware design. Especially if a slim Arch Linux is running on it. I only can recommend it. If you have any problems setting it up, don't hesitate to contact me :) Ubuntu, Debian... will probably fail, due to old kernels and old software, which still don't include all the patches... I would suggest Arch Linux, especially because the very first OpenRazer releases run flawless on this system. Otherwise, please feel free to help porting it to other Linux Distributions.

proxymus89, 2016-08-24:
@m4ng0squ4sh Thank you so much for your support!!!When i will buy my razer i will contact you sure :) (before buy i have to sell my macbook pro)

> I had a lot of troubles but got everything fixed now. Systemd-Boot freeze, Kernel problems, GPU stuff, DSDT firmware fix, LED lighting modes

You had those problems with Ubuntu/Debian or with Arch linux?My idea is to install Kali linux(which is based on debian), but i could try to install Black arch if is more supported

> I really love the Razer Blade 14 and it's a masterpiece of hardware design. Especially if a slim Arch Linux is running on it. I only can recommend it.

I'm totally agree with you :)

> If you have any problems setting it up, don't hesitate to contact me 

Thanks, i will do :)

m4ng0squ4sh, 2016-08-24:
@proxymus89 I only run Debian/CentOS on servers... All Debian based systems will have a problem with the new Razer Blade, except if you build the latest kernel and apply the patches manually. Just give Arch a try or wait until Debian,... update their packages. ;)

Gorta, 2016-08-24:
Hi, 
I take advantage of this topic for a little question.
Do you will add support of razer gamepad (orbweaver and tartarus) ? (maybe they are already supported but it's not write in the readme ^^).

Thank for your good job.

terrycain, 2016-08-24:
Tartarus is supported [here](https://github.com/terrycain/razer_drivers). Orbweaver isnt supported but shouldnt be difficult.

Gorta, 2016-08-24:
Thank you ;)
Do you need information for add support of Orbeweaver ?

terrycain, 2016-08-24:
Yeah that would be good :smile: 

If I'm not mistaken the Orbweaver is a suped up version of the Tartarus. I have all of the Tartarus working so that should apply to the Orbweaver but I would need examples of the individual LED lighting bits. If you think your capable of [this](https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol) then go [here](https://github.com/terrycain/razer_drivers) and make a new issue for Orbweaver support and we'll get cracking.

Gorta, 2016-08-24:
Ok. I will look that tomorrow. 
Many thanks ! :)

terrycain, 2016-08-24:
Np