pez2001/razer_chroma_drivers #158: razer-service: No module named 'notify2'

kusayuzayushko, 2016-08-28:
Can't run the daemon on Archlinux.

```
$ razer-service                                                                                    
Traceback (most recent call last):
  File "/usr/bin/razer-service", line 6, in <module>
    from razer_daemon.daemon import daemonize
  File "/usr/lib/python3.5/site-packages/razer_daemon/daemon.py", line 23, in <module>
    import razer_daemon.hardware
  File "/usr/lib/python3.5/site-packages/razer_daemon/hardware/__init__.py", line 7, in <module>
    from razer_daemon.hardware.mouse import RazerMambaChromaWireless
  File "/usr/lib/python3.5/site-packages/razer_daemon/hardware/mouse.py", line 5, in <module>
    from razer_daemon.misc.battery_notifier import BatteryManager
  File "/usr/lib/python3.5/site-packages/razer_daemon/misc/battery_notifier.py", line 8, in <module>
    import notify2
ImportError: No module named 'notify2'
```

terrycain, 2016-08-28:
Should of raised the issue [here](https://github.com/terrycain/razer-drivers) :wink:. Install (asuming your on Ubuntu) `apt-get install -y python3-notify2`.

Has been fixed ready for the next release.

closed 2016-08-28 by terrycain

kusayuzayushko, 2016-08-29:
Yep, installing `python-notify2` from AUR solved this.

miracode2033, 2019-01-04:
Installed, it declines to work.
[image]