pez2001/razer_chroma_drivers #161: Errors loading modules on Debian Unstable

CalcProgrammer1, 2016-12-19:
I'm trying to install this on Debian Unstable (sid) with kernel version 4.8.0-2-amd64.  I created the Debian package using the script and dpkg installed it.  The DKMS system built the module successfully.  However, when I load any of the modules I get missing symbols errors in dmesg:

`[ 2111.612776] razerkbd: Unknown symbol razer_chroma_extended_matrix_effect_breathing_random (err 0)
[ 2111.612784] razerkbd: Unknown symbol razer_chroma_standard_matrix_effect_breathing_random (err 0)
[ 2111.612792] razerkbd: Unknown symbol razer_chroma_extended_matrix_effect_breathing_dual (err 0)
[ 2111.612808] razerkbd: Unknown symbol razer_chroma_standard_matrix_effect_custom_frame (err 0)
[ 2111.612816] razerkbd: Unknown symbol razer_chroma_standard_set_led_brightness (err 0)
[ 2111.612823] razerkbd: Unknown symbol razer_chroma_standard_set_device_mode (err 0)
[ 2111.612838] razerkbd: Unknown symbol razer_chroma_standard_get_led_state (err 0)
[ 2111.612845] razerkbd: Unknown symbol razer_chroma_standard_matrix_effect_wave (err 0)
[ 2111.612852] razerkbd: Unknown symbol razer_chroma_standard_get_led_effect (err 0)
[ 2111.612860] razerkbd: Unknown symbol razer_chroma_extended_matrix_effect_reactive (err 0)
[ 2111.612868] razerkbd: Unknown symbol razer_chroma_standard_matrix_effect_none (err 0)
[ 2111.612876] razerkbd: Unknown symbol razer_chroma_standard_get_led_brightness (err 0)
[ 2111.612884] razerkbd: Unknown symbol razer_chroma_standard_get_firmware_version (err 0)
[ 2111.612891] razerkbd: Unknown symbol razer_chroma_extended_matrix_effect_static (err 0)
[ 2111.612899] razerkbd: Unknown symbol razer_chroma_standard_set_led_state (err 0)
[ 2111.612907] razerkbd: Unknown symbol razer_chroma_standard_matrix_effect_spectrum (err 0)
[ 2111.612922] razerkbd: Unknown symbol razer_chroma_standard_get_device_mode (err 0)
[ 2111.612929] razerkbd: Unknown symbol razer_chroma_extended_matrix_set_custom_frame (err 0)
[ 2111.612937] razerkbd: Unknown symbol razer_chroma_extended_matrix_effect_breathing_single (err 0)
[ 2111.612945] razerkbd: Unknown symbol razer_chroma_misc_fn_key_toggle (err 0)
[ 2111.612953] razerkbd: Unknown symbol razer_chroma_extended_matrix_effect_none (err 0)
[ 2111.612960] razerkbd: Unknown symbol razer_chroma_extended_matrix_effect_custom_frame (err 0)
[ 2111.612968] razerkbd: Unknown symbol razer_chroma_standard_get_serial (err 0)
[ 2111.612975] razerkbd: Unknown symbol razer_chroma_extended_matrix_effect_wave (err 0)
[ 2111.612999] razerkbd: Unknown symbol razer_chroma_standard_matrix_set_custom_frame (err 0)
[ 2111.613007] razerkbd: Unknown symbol razer_chroma_misc_get_blade_brightness (err 0)
[ 2111.613014] razerkbd: Unknown symbol razer_chroma_extended_matrix_effect_starlight_single (err 0)
[ 2111.613022] razerkbd: Unknown symbol razer_chroma_standard_matrix_effect_breathing_single (err 0)
[ 2111.613030] razerkbd: Unknown symbol razer_chroma_extended_matrix_effect_starlight_dual (err 0)
[ 2111.613037] razerkbd: Unknown symbol razer_chroma_misc_set_blade_brightness (err 0)
[ 2111.613051] razerkbd: Unknown symbol razer_chroma_extended_matrix_brightness (err 0)
[ 2111.613059] razerkbd: Unknown symbol razer_chroma_standard_set_led_effect (err 0)
[ 2111.613067] razerkbd: Unknown symbol razer_chroma_extended_matrix_effect_spectrum (err 0)
[ 2111.613075] razerkbd: Unknown symbol razer_chroma_standard_matrix_effect_static (err 0)
[ 2111.613082] razerkbd: Unknown symbol razer_chroma_extended_matrix_effect_starlight_random (err 0)
[ 2111.613095] razerkbd: Unknown symbol razer_chroma_standard_matrix_effect_breathing_dual (err 0)
[ 2111.613103] razerkbd: Unknown symbol razer_chroma_standard_matrix_effect_reactive (err 0)
[ 2111.613112] razerkbd: Unknown symbol razer_chroma_standard_matrix_effect_starlight_single (err 0)
`
Then I'm trying to integrate this into my application (porting my Keyboard Visualizer to Linux, going to replace the official Chroma SDK with this project) and I call "chroma = razer_open(NULL, NULL)" and the program exits immediately rather than returning NULL.  I would not like my program to close if the Chroma library fails to initialize, because I wish to support more than just Chroma devices.  I am linking against librazerchroma.a.

What am I doing wrong?

terrycain, 2016-12-19:
Hmm probs dkms didn't build it properly. Will have a look tonight if I get a chance. 

Inside the repo itself run `make driver_verbose` and then try `sudo insmod driver/razerkbd.ko`

CalcProgrammer1, 2016-12-19:
Your fork worked for me, so it's something with this repository.

terrycain, 2016-12-19:
Hmm, i could of broke things ;) will have a look once i've finished murdering udev