pez2001/razer_chroma_drivers #170: Razer Keyboard Driver - Linux

whiteknights2007, 2017-10-08:
I followed the steps to install the drivers for razer products from this website (https://openrazer.github.io/) , and installed all drivers for the polychromatic controller from this website (http://www.omgubuntu.co.uk/2016/09/polychromatic-new-linux-razer-drivers) . The problem is that I cant probe the keyboard driver . I following this procedure (https://github.com/openrazer/openrazer/wiki/Setting-up-the-keyboard-driver) , but it still didnt work. all other softwares (like razerCommander and razerGenie) dont work , so I'm pretty sure the problem is setting up the keyboard driver. 

Please help!!! 

(Im using UBUNTU 16.04 LTS)

trying modprobe razerkbd gives : 
Blade-Stealth:~$ modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Operation not permitted

lah7, 2017-10-08:
This is the wrong repo, you should raise issues for OpenRazer on https://github.com/openrazer/openrazer instead. You may now have the wrong package/repo installed as that article for Polychromatic is outdated.

[See these instructions instead](https://github.com/lah7/polychromatic#driver-installation) to install both properly.

whiteknights2007, 2017-10-08:
Thanks for the fast response. 
I tried the commands on the website you sent me and this is what I see now :  

modprobe razerkbd
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.10.0-35-generic